avanc/photOS #6: Add animated GIF Support

escuco, 2021-01-11:
It would be nice, if PhotOS would support animated GIF files. Thus, for example, 'video' loops and cinemagraphs could be played on the photo frame.
Examples: 
http://cinemagraphs.com/
https://www.gallereplay.com/

avanc, 2021-01-18:
* [giflib](http://giflib.sourceforge.net/)
    * was used by fbv
    * not sure if it supports animations: http://giflib.sourceforge.net/whatsinagif/animation_and_transparency.html

escuco, 2021-02-18:
Did some testing with graphic output (spec. animated gif) on framebuffer:

fbi -> only shows the first frame from the animated gif
fim -> only shows the first frame from the animated gif
feh -> doesn't support framebuffer

gif_play -> got stuck while compiling due to missing dependencies and other error messages ¯\_(ツ)_/¯

escuco, 2021-02-18:
Maybe playing cinemagraph video files is an option instead of animated gifs:

All players are tested with mp4 no audio cinemagraphs with 1920x1080 pixel resolution

[mplayer](https://mplayerhq.hu/) -> to slow on raspberry pi zero

[vlc](https://www.videolan.org/vlc/) -> works nice
`vlc --no-video-title --fbdev /dev/fb1 filename.mp4`

[omxplayer](https://github.com/popcornmix/omxplayer) (command line video player for raspberry pi) -> works nice
`omxplayer -b filename.mp4`

Schrigga, 2021-08-07:
I have created a pull request which adds video support

escuco, 2021-12-04:
So from my point of view, this issue could be closed. The video support works quite well and with it, there's no longer need for gif support.

avanc, 2021-12-04:
That's true.

closed 2021-12-04 by avanc