avanc/photOS #11: Rotate images based on metadata

avanc, 2021-02-08:
Some images (mainly from smartphones) contain metadata to rotate the image:
https://www.impulseadventure.com/photo/exif-orientation.html

This is currently not supported by fbv and images are shown wrongly.

kmuehlbauer, 2021-05-10:
@avanc Would you consider adding `ImageMagick` to the image? Then we could just rotate the image when syncing using convert/mogrify.

avanc, 2021-05-10:
@kmuehlbauer Sounds like a good workaround.
Initially, I was afraid that those locally changed files are overwritten on each sync. But it seems that `--ignore-existing` will do the trick: https://askubuntu.com/questions/601878/rsync-delete-files-from-target-if-not-existing-in-source

What would be the imagemagick command? Can it read EXIF information?

kmuehlbauer, 2021-05-10:
This will do the inplace rotation:

```bash
$ mogrify -auto-orient your-image.jpg
```

avanc, 2021-05-10:
I just realized, that mogrify does also change the file if the image is already rotated. Thus, every image is rewritten after each sync, which should be avoided.

I used exiftran on my PC which only changes files which have to be rotated. However, it has a lot of dependencies :-/

kmuehlbauer, 2021-05-10:
> I just realized, that mogrify does also change the file if the image is already rotated.

Maybe because it strips the EXIF information from the file? But doesn't `--ignore-existing` cover for that? Another problem might arise, if the EXIF information is wrong.

kmuehlbauer, 2021-05-14:
@avanc I've compiled `photOS` from scratch with `ImageMagick`. I'm working around the issue with changed files by not changing them.

What I do instead is, converting the image into the `/tmp`-folder and feed it to `fbv` from there. Afterwards I delete this file.

This is the start-function in `/usr/bin/photoframe.sh` with the needed changes

```bash
function start {                                                                       
  counter=0                                                          
  error_settopic 01_Startup                                                               
  error_write "Go to http://$(hostname) to configure photOS"                           
                                                                     
  while true; do                                                                          
    IMAGE=$(get_image)                                                                 
    echo $IMAGE                                                      
                                                                                          
    # convert-rotate image to /tmp-folder                                              
    IMAGE2=/tmp/$(basename $IMAGE)                                   
    convert -auto-orient "$IMAGE" "$IMAGE2"                                               
    fbv $PARAMS_FBV "$IMAGE2"                                                          
    error_display                                                    
    sleep $SLIDESHOW_DELAY                                                                
    rm "$IMAGE2"                                                                       
                                                                     
    counter=$((counter+1))                                                                
    if [ $counter -eq 10 ]                                                             
    then                                                             
      error_settopic 01_Startup                                                           
    fi                                                                                 
                                                                     
  done                                                                                    
}
```

This works actually quite good in my test cases. 

Not sure, if a clean-up script which handles stale images in `/tmp` might be needed, if the image isn't removed for some reason.

avanc, 2021-05-15:
Good proposal. It should be possible to _hardcode_ image2 filename. Than we don't have to care about removing it...

Still curious what the difference between convert and mogrify is.

kmuehlbauer, 2021-05-15:
Yes, hardcoding should work better. I haven't thought about that.

`convert` does conversion from input-file to output-file while `mogrify` does the conversion in-place (changing the input file).

avanc, 2021-05-20:
Your approach is now available in the latest release:
https://github.com/avanc/photOS/releases/tag/v0.3.2

Thanks @kmuehlbauer

closed 2021-05-20 by avanc

TuxMaus, 2024-01-16:
In my installation (V 0.3.7), the (same) rotatet image is saved in /tmp very often (today: 62 times within 17 hours - see below). But the files are never deleted and due to this /tmp runs out of memory:

```
[root@photos-65de0386 /]# ls -lt /tmp | grep "1637352 Jan 16"
-rw-r--r--    1 root     root       1637352 Jan 16 16:48 vVmfwi
-rw-r--r--    1 root     root       1637352 Jan 16 16:45 hJfWBU
-rw-r--r--    1 root     root       1637352 Jan 16 16:42 JsmgJJ
-rw-r--r--    1 root     root       1637352 Jan 16 16:32 UyOBfb
-rw-r--r--    1 root     root       1637352 Jan 16 16:29 WG0DlI
-rw-r--r--    1 root     root       1637352 Jan 16 16:17 W7Bvuh
-rw-r--r--    1 root     root       1637352 Jan 16 16:07 0jnqdG
-rw-r--r--    1 root     root       1637352 Jan 16 15:37 D4JGBh
-rw-r--r--    1 root     root       1637352 Jan 16 14:47 ZofcmL
-rw-r--r--    1 root     root       1637352 Jan 16 14:40 X1ltkQ
-rw-r--r--    1 root     root       1637352 Jan 16 14:16 PqQwsL
-rw-r--r--    1 root     root       1637352 Jan 16 14:12 KtTSma
-rw-r--r--    1 root     root       1637352 Jan 16 13:55 guotWv
-rw-r--r--    1 root     root       1637352 Jan 16 13:20 Lm2EOE
-rw-r--r--    1 root     root       1637352 Jan 16 13:16 KbwDlm
-rw-r--r--    1 root     root       1637352 Jan 16 13:15 1XfJVz
-rw-r--r--    1 root     root       1637352 Jan 16 13:02 MTpOX1
-rw-r--r--    1 root     root       1637352 Jan 16 12:17 vEOfxS
-rw-r--r--    1 root     root       1637352 Jan 16 12:16 lbAlVS
-rw-r--r--    1 root     root       1637352 Jan 16 11:51 MzdwF3
-rw-r--r--    1 root     root       1637352 Jan 16 10:29 k80OYP
-rw-r--r--    1 root     root       1637352 Jan 16 09:54 F6ALE2
-rw-r--r--    1 root     root       1637352 Jan 16 09:45 oXW8QT
-rw-r--r--    1 root     root       1637352 Jan 16 09:30 3nZHwy
-rw-r--r--    1 root     root       1637352 Jan 16 09:10 StXZIs
-rw-r--r--    1 root     root       1637352 Jan 16 08:41 eb2E3B
-rw-r--r--    1 root     root       1637352 Jan 16 08:35 bUUHbM
-rw-r--r--    1 root     root       1637352 Jan 16 08:15 SlJcON
-rw-r--r--    1 root     root       1637352 Jan 16 08:07 PpKAAb
-rw-r--r--    1 root     root       1637352 Jan 16 07:55 fAvlbn
-rw-r--r--    1 root     root       1637352 Jan 16 07:38 7diWRD
-rw-r--r--    1 root     root       1637352 Jan 16 07:26 b7okqL
-rw-r--r--    1 root     root       1637352 Jan 16 07:23 7iCn59
-rw-r--r--    1 root     root       1637352 Jan 16 07:07 Z9FON6
-rw-r--r--    1 root     root       1637352 Jan 16 07:03 xqK0Hh
-rw-r--r--    1 root     root       1637352 Jan 16 06:53 VKHK0F
-rw-r--r--    1 root     root       1637352 Jan 16 06:51 EHH1oo
-rw-r--r--    1 root     root       1637352 Jan 16 06:49 ECWOiu
-rw-r--r--    1 root     root       1637352 Jan 16 06:48 fYMHld
-rw-r--r--    1 root     root       1637352 Jan 16 06:41 5NgwSb
-rw-r--r--    1 root     root       1637352 Jan 16 06:34 svAtSc
-rw-r--r--    1 root     root       1637352 Jan 16 06:27 4Z0P9b
-rw-r--r--    1 root     root       1637352 Jan 16 05:42 99TocE
-rw-r--r--    1 root     root       1637352 Jan 16 05:39 A4yg3O
-rw-r--r--    1 root     root       1637352 Jan 16 05:35 WqwMnT
-rw-r--r--    1 root     root       1637352 Jan 16 04:54 AF44Li
-rw-r--r--    1 root     root       1637352 Jan 16 04:07 tH89b5
-rw-r--r--    1 root     root       1637352 Jan 16 03:55 SQATGQ
-rw-r--r--    1 root     root       1637352 Jan 16 03:45 YoobzD
-rw-r--r--    1 root     root       1637352 Jan 16 03:10 77zQ2G
-rw-r--r--    1 root     root       1637352 Jan 16 02:39 4QGXYQ
-rw-r--r--    1 root     root       1637352 Jan 16 02:29 HdEfyH
-rw-r--r--    1 root     root       1637352 Jan 16 02:16 WHj0eI
-rw-r--r--    1 root     root       1637352 Jan 16 01:57 GIETp7
-rw-r--r--    1 root     root       1637352 Jan 16 01:22 I713Qj
-rw-r--r--    1 root     root       1637352 Jan 16 01:02 IEqWnJ
-rw-r--r--    1 root     root       1637352 Jan 16 00:57 TZoQLb
-rw-r--r--    1 root     root       1637352 Jan 16 00:55 hiYAf1
-rw-r--r--    1 root     root       1637352 Jan 16 00:47 pEkTAG
-rw-r--r--    1 root     root       1637352 Jan 16 00:37 yUy3xc
-rw-r--r--    1 root     root       1637352 Jan 16 00:09 Zebfsf
-rw-r--r--    1 root     root       1637352 Jan 16 00:00 slYmV4
```

TuxMaus, 2024-01-17:
You guessed it right: It's a Problem with jhead and this specific picture due to (probalby) incorrect EXIF data (and therefore not a bug of PhotOS). 

The temporary file is created but not deleted due to an error:
```
[root@photos-65de0386 tmp]# jhead -autorot /tmp/2024-01-06_20231230_092728.jpg 
Invalid SOS parameters for sequential JPEG
Error : Problem executing specified command
in file '/tmp/2024-01-06_20231230_092728.jpg'

[root@photos-65de0386 tmp]# ls -la
-rw-------    1 root     root       1653902 Jan 17 16:31 2024-01-06_20231230_092728.jpg
-rw-------    1 root     root       1637352 Jan 17 16:32 BYduLw
```
Since this is a problem that only affects this one file with incorrect EXIF data, I don't think further action is necessary. Perhaps I will try to program a cronjob that monitors the /tmp directory and deletes old temporary jhead files from time to time...

Thanks for the awesome support!