avanc/photOS #18: Does not work

knumsi, 2021-04-05:
Just leaving a Comment. I like your basic principle using webdav as source and configuration done via web.

But I do not get it running. RPI3 7" Touchscreen + RPI3.

System booting, SSH open and root login without any password possible. No webinterface though it shows to configure webdav server ...

Therefor I leave searching for another solution.

avanc, 2021-04-07:
Sorry for the problems. I never tried photOS using the DSI port. Did you get an output on the display?

Nevertheless, the web interface should be available also if no display is connected at all.
I'll try to reproduce your problem with a RPI3 without any display the next days.

avanc, 2021-04-08:
It was a stupid bug in the package configuration.
I grabbed a Raspberry Pi 3B+ to test it. In addition, a Raspberry Pi 4 is ordered, so I can test the software on all devices before doing a release.

@knumsi apologize for the inconvenience

closed 2021-04-08 by avanc

avanc, 2021-04-09:
@knumsi A new release is available which should solve your problem: https://github.com/avanc/photOS/releases/tag/v0.2.2

However, I'm still not sure if the DSI-Display will work out of the box. From some tutorials I read it seems that nothing has to be done. But I'm not sure if this is only valid for Rasbian based images...
Any feedback is highly welcome.

commit 3be2891e (2021-04-08): Add missing package dpendencies

Fixes #18