avanc/photOS #20: Webserver not responding

cbartetzko, 2021-04-08:
I've installed latest realease on Raspi Zero, Raspi 2, Raspi 3, each with respective image.
However the result is always the same:
Installation goes well, I get to the photoOS WelcomeScreen.
When trying to access the Configuration with a webbrowser I get no connection.
Chrome says: ERR_CONNECTION_REFUSED
Accessing by ssh or ping works without problems.

avanc, 2021-04-08:
Unfortunately, there is a bug for RPI2, RPI3 and RPI4: https://github.com/avanc/photOS/issues/18
I'm currently creating a new release to solve the issue.

However, having the same problem with RPI zero using the photOS-raspberrypi-0.2.1.img.xz image is strange.
Can you see any error messages on the display while the system is booting?

avanc, 2021-04-08:
@cbartetzko You may want to give the new release a try?
https://github.com/avanc/photOS/releases/tag/v0.2.2

avanc, 2021-04-30:
@cbartetzko Could you reproduce the problem with the latest release? Otherwise, I would close this issue.

cbartetzko, 2021-04-30:
You can close this issue. It's working fine now. Thanks.

avanc, 2021-04-30:
Great

closed 2021-04-30 by avanc