avanc/photOS #21: Support for motion sensors to switch off/on HDMI (display)

KarlR83, 2021-04-29:
Great project, also the MAKE article about it!

You already implemented time schedule based switch off/on of the HDMI port (and of the connected display), which is nice.

There are also motions sensors out there ([example](https://maker.pro/raspberry-pi/tutorial/how-to-interface-a-pir-motion-sensor-with-raspberry-pi-gpio)) that could be used to switch the display off (after a grace period) and on (immediatly) depending on presence (motion dectected) in the room.
In fact I use on of these and a shell script to do so. Not sure how to implement this in the configuration webpage, though (which would be great). Maybe you can work on this in the future. I could provide my (very simple) shell scripts. if this helps.

avanc, 2021-04-30:
@KarlR83 Sounds like a very good idea!

I did something similar with motioneyeos: https://github.com/avanc/motioneye-pir
I cann recommend AM312 PIR sensors. They have a good detection range and are very small.

However, in my compact hardware setup I had problems that the wifi was influencing the PIR sensor. So make sure they are not close together.

I'll take this as enhancement for a next release.

avanc, 2021-05-07:
@KarlR83 Would be great if you could test this release:
https://github.com/avanc/photOS/releases/tag/v0.x.y-pirtest

This includes an additional service to read a GPIO and activate/deactivate the display accordingly. There is no webui, you have to do the configuration via ssh and add the file `/data/photoframe/motion_detector.conf` with the following content:
https://github.com/avanc/photOS/blob/pir/package/photoframe/pir/motion_detector.conf

GPIO is the RPi GIO where the sensor is connected.
COMMAND_ON/OFF is the command to be executed if motion is detected (in this example activating/deactivating the display)

In addition you can set:
INTERNAL_RESISTOR=on (if your sensor is floating, or in my case if a button is connected to GND without external pull-up resistor)
STOP_DELAY=10 (seconds after last motion is detected when the display is switched off, 10 seconds by default for testing)
MAX_LENGTH=600 (overall active seconds, the display is deactivate if the display for that amount of time also if there is still motion, feature deactivated by default: MAX_LENGTH=0)

Any feedback is highly welcome.

KarlR83, 2021-05-08:
Great, thanks a lot. Not sure if I will manage to test this in the next days, but I will for sure try it out and let you know.

KarlR83, 2021-05-11:
Some first feedback:
Setting display on/off times results in server error 500. Changing the timezone doesn't work (always resets to default "Africa/Abidjan".
But maybe that's expected due to the changes.
Will get back to you, once I tested the sensor.

avanc, 2021-05-11:
There was a bug regarding display schedule: https://github.com/avanc/photOS/issues/27
This PIR prerelease based on the old version.

KarlR83, 2021-05-12:
Tested it yesterday, but seems it isn't working for me. The display never goes off.

My sensor is connected to GPIO=23
I also tried with STOP_DELAY=3 and INTERNAL_RESISTOR=on (not sure if my sensor is floating, though) but still doesn't work.

Any way how to switch on debugging mode, so I can for example get log output to see if motion is detected or not?

avanc, 2021-05-12:
You are right, I added a wrong path to the configuration path.
I'm building a new test image:
https://github.com/avanc/photOS/actions/runs/835570175

I let you know when it is available.

avanc, 2021-05-13:
@KarlR83 New pre-release is now available:
https://github.com/avanc/photOS/releases/tag/v0.x.y-pirtest3

Configuration file path is now: `/data/photoframe/conf/motion_detector.conf`

For testing, stop the service and start the script by hand:

    /etc/init.d/S81motiondetector stop
    /usr/bin/motion_detector.py

This should give you some debugging output.

KarlR83, 2021-05-13:
@avanc Works. great work!
(Seems my sensor is is not floating, didn't try the INTERNAL_RESISTOR option)

Changing the timezone using the UI still doesn't work (always resets to default "Africa/Abidjan"). Or is this just an UI issue?

Anyway, would be great to see the motion detection feature in the next release!

avanc, 2021-05-13:
Glad to hear that it is working. I will include it in the next release with some documentation in the wiki.

Unfortunately, I cannot reproduce your problem with the timezone. Please check if there are files with ending `.py` (don't care about `.pyc`) under  `/data/usr/lib/python3.8/site-packages/webui/`.

You can also stop the webui service and start it manually to see some possible errors:

    /etc/init.d/S81webui stop
    /usr/bin/thingos_webui start

If the problem still exist, please open a new issue.

KarlR83, 2021-05-14:
There are no .py files in that directory, see screenshot.
[image]

Manually stopping and starting the webui does not give me any errors (on console). Setting the timezone still doesn't work.

I will wait for the next release. If I still face this problem, I will open an issue.

avanc, 2021-05-15:
@KarlR83 One hint from a friend: Note that the drop-down menu will always show the first timezone. The selected timezone is printed above...

KarlR83, 2021-05-15:
That's it, sorry for the confusion. I was just looking at the drop-down, not the text above it. So no issue here.

closed 2021-05-15 by KarlR83

avanc, 2021-05-17:
> That's it, sorry for the confusion. I was just looking at the drop-down, not the text above it. So no issue here.

I'll update the webui in the next release to avoid this confusion.

KarlR83, 2021-05-21:
Works great with the new 0.3.2 release, thanks again!

As a (really minor) hint, at the [Motion Detection Wiki](https://github.com/avanc/photOS/wiki/Motion-Detection) you say
`vim /data/photoframe/conf/motion_detector.conf`
But there is no _vim_ in photOS, only _vi_

avanc, 2021-05-21:
Thanks for the hint!