avanc/photOS #25: why only webdav?

mrlaja, 2021-05-06:
the Project is perfect! release v0.3.0-rc1 works fine.

id would be perfect to connect the Pictureframe via smb or sftp.
my Openmediavault 5 server do not WebDav...

avanc, 2021-05-06:
I already had a Nextcloud server when I started the project. Thus, webdav was the obvious choice at that time.
But I agree that smb and sftp are both good alternatives.
What would be your favourite?

mrlaja, 2021-05-06:
Smb would be the Favorite for the User in their own Networks. Butt sftp ist better for User in other Networks. I would Like to Install such an pictureframe at my parents place and host the Pictures on my Server. So they Always can See Always nee Images Form their Grandchildren..... My moms Smartphone makes Backups in my Server, and IT works fine.

So stfp.

⁣mit freundlichen Grüßen
Jan Lange

chr2k, 2021-05-12:
Great if smb or similar would be integrated :-)

admindonkey, 2021-07-22:
Hi,
does enybody know how I can realice the SMB sync?

Greetings

avanc, 2021-07-22:
@admindonkey Unfortunately, SMB is not supported yet. I already had a look in different tools, which might be used for SMB and maybe also other protocols. But not ready.

admindonkey, 2021-07-22:
Oh, how pity. 
Do you think it will be possible in near future?

avanc, 2021-07-23:
I plan to use [Rclone](https://rclone.org/) as it would allow a bunch of sync sources (e.g. SMB, SFTP, WebDAV). However, there is no package for buildroot and I have to create it. As I'm not that familiar with go-lang, I don't want to promise anything. But I'm optimistic to have SMB supported within the holiday season :-)

FriendOfphotoOS, 2023-10-11:
Hi, is there already a date for when the smb protocol will be integrated into photoOS?