avanc/photOS #27: Automatic display set fail

UdoKraushaar, 2021-05-10:
Hi
Setting the automatic display activation times (e.g. on 08:00 , off 18:00) in the web GUI results in an internal server error 500.

In photoframe.conf only entry is SLIDESHOW_DELAY=_'time-set-by-me'_. File is rw for root.
Until possible fix: what should be the entry in the conf file for above mentioned time example?
Thank you

avanc, 2021-05-10:
@UdoKraushaar I can reproduce the problem and investigating. Thanks for reporting.

As workaround: The times are stored in `/data/etc/crontabs/root`. Example for turning on at 7am, and switching of at 9pm:

    0 7 * * * /usr/bin/photoframe.sh display on
    0 21 * * * /usr/bin/photoframe.sh display off

avanc, 2021-05-10:
New release will be available in several hours.

avanc, 2021-05-10:
Bugfix is released: https://github.com/avanc/photOS/releases/tag/v0.3.1

commit 96ac22fd (2021-05-10): Update webui

Fixes #27