avanc/photOS #33: Unable to access file or unknown file format

makuris, 2021-05-21:
Hi, i have updated to 0.3.2 and now from time to time I get the following new error message:
[image]

New added images are now shown in the correct rotation. But why do I get the error message?

avanc, 2021-05-21:
It seems that the convert command is killed maybe due to memory limitations or it takes to long. Thereafter, fbv (frame buffer viewer) tries to show the converted image but it is not there or broken...

How often does this happen and how many images do you have?

If it happens because ImageMagicks convert cannot convert one specific file, you should be able to find it by deactivating shuffle (https://github.com/avanc/photOS/wiki/FAQ#how-do-i-deactivate-photo-shuffling) and see which file is never shown.

But I don't have a real clue yet...

makuris, 2021-05-22:
The error message appears every 10 images or so. I have currently 250 images on the webdav folder. I rotated most of them using irfanview since the image magic trick was not implemented at the beginning. 

The raspi is capable of showing really large images >15MB if he does not need to rotate them.

So there are about 25 portrait images that need rotating. I went through and I think they were never shown, so I guess these images cause the error. They were all taken with a smartphone and file size varies between 250KB and 2MB.
I use a raspi zero, so what would be the max. Image size regarding the available memory?

avanc, 2021-05-22:
Thanks for the details. I'll take a look into it after the weekend.

avanc, 2021-05-25:
@makuris  Just to let you know that I can reproduce the problem, but still investigating.

avanc, 2021-05-26:
@makuris Can you give the prerelease a try: https://github.com/avanc/photOS/releases/tag/v0.3.3-rc1
This should solve the problem.

ThorstenEisinger, 2021-05-26:
Seems to solve the problem for me with a Raspberry Pi Zero W. Thanks for the quick fix.

makuris, 2021-05-26:
It now rotates all images correctly and no more error messages. This is awesome, thank you so much!

avanc, 2021-05-28:
@makuris and @ThorstenEisinger : Thanks for the quick feedback.

I just released 0.3.3 with the fix of the release candidate.

closed 2021-05-28 by avanc