avanc/photOS #34: How to (temporarily) disable the read-only filesystem?

alex0124000, 2021-05-22:
I have a local webdav server (a Synology NAS) which uses HTTPS but has a self-signed certificate.
photOS (mount.davfs) refuses to speak to the server because the certificate can't be validated.

According to the mount.davfs documentation (https://linux.die.net/man/8/mount.davfs), I could copy the certificate to /etc/davfs2/certs or ~/.davfs2/certs but both locations are read-only (when accessing the photOS machine over ssh).

Can I somehow (on a windows machine) enable/disable the read-only protection? When I stick the SD card in my windows machine, I can read only the first of the three partitions.

avanc, 2021-05-22:
The command is 

    mount -o remount,rw /

However, the changes will be overwritten after update.

Would be great if you could write your steps, as I heard about the problem before but didn't know how this can be done. I would add the details in the wiki.

avanc, 2021-05-22:
I'll think how to add certificates permanently.

alex0124000, 2021-05-22:
Thanks for your help!
I got it working now.
These are the necessary steps:
```
mount -o remount,rw /
mkdir -p /etc/davfs2/certs/
vi /etc/davfs2/certs/ownCert.pem
# I added only my own self-signed certificate here (starting with line -----BEGIN CERTIFICATE-----)
# I think I could have chosen any other file path since I need to add an explicit reference in the webdav.conf anyway.

vi /etc/photoframe/davfs2.conf
#change the file such that it contains these two lines (without #)
#secrets /data/photoframe/conf/webdav.conf
#trust_server_cert /etc/davfs2/certs/ownCert.pem
```

The changes survive a reconfiguration of the webdav settings (via web UI), but they are overwritten on a server restart (as you already wrote).

MeWorkstar, 2021-05-31:
I have a similar problem / setup and i am interested in a solution.
Can't wait to see your approach and the documentation.

If help or more infos are needed from my case, feel free to reach out to me.

avanc, 2021-05-31:
@MeWorkstar No progress yet, sorry. However, @alex0124000 approach https://github.com/avanc/photOS/issues/34#issuecomment-846457203 should work.

You should store webdav_cert.pem somewhere to /data (e.g. /data/photoframe/conf/webdav_cert.pem) so it is not removed on an update. However, you have to do the changes to /etc/photoframe/davfs2.conf after each update.

alex0124000, 2021-06-13:
thanks for merging my pull request. Shall we close this issue now?

avanc, 2021-06-13:
Self signed certificates is not included yet. I want to keep this open until I also updated the documentation.

alex0124000, 2021-06-13:
Ah, you are right. I got confused by the two topics.

closed 2021-06-22 by avanc

avanc, 2021-06-22:
@alex0124000 It is now included in this release candidate: https://github.com/avanc/photOS/releases/tag/v0.3.4-rc1
Maybe you want to give it a try.
Its documented at https://github.com/avanc/photOS/wiki/Self-signed-certificate

alex0124000, 2021-06-27:
I tried `https://github.com/avanc/photOS/releases/tag/v0.3.4-rc2` (the files are still named `...rc1` :) )
I ran into one problem: I got this error message when it tried to connect to my NAS.

> Mounting .../my_photos/photoframe failed: mount.davfs: can't read server certificate /data/photoframe/conf/webdav_cert.pem

I think it was caused by the file permissions on the .pem, but I forgot to confirm what the current settings were.
I then ran `chmod a+r /data/photoframe/conf/webdav_cert.pem` and the connection worked. So I think this was the problem.
Maybe the scp command overwrote the file (including its permission settings)? We could just add the `chmod` to `photoframe.sh`.

commit 69c3d122 (2021-06-22): Add support for self-signed certificates

Fixes #34