avanc/photOS #38: Permission denied on MacOS

steamfan, 2021-06-08:
I'm trying to write the image to the micro SD card with the writeimage.sh script on MacOS and get a permission denied. Following commands:

> ls
photOS-raspberrypi-0.3.3.img    photOS-raspberrypi-0.3.3.img.xz writeimage.sh

According to DiskUtility.app shows to Card Reader is 'disk3'; <MeinWLAN>:<MeinWLANPassword> are replaced here.

> sudo ./writeimage.sh -d /dev/disk3 -i "/Users/homsch/Pictures/20090809_NorthStradbrokeIsland_087_klein.jpg" -n '<MeinWLAN>:<MeinWLANPassword>'

>  * writing disk image to sdcard
1+1 records in
1+1 records out
1913676 bytes transferred in 0.260613 secs (7342981 bytes/sec)
 * mounting sdcard
mkdir: ./.boot: Permission denied

Any help is appreciated

avanc, 2021-06-09:
It seems that writing the image works. As workaround you can insert the SD-Card again and mount the first partition (should happen automatically) to configure the network (https://github.com/avanc/photOS/wiki/Installation Step 4).

It seems that the new partitions are not named disk3s1..3. Can you have a look how MacOS names them (`ls /dev/disk3*`)

avanc, 2021-06-09:
@escuco Am I right that you had a similar problem?

escuco, 2021-06-10:
@avanc @steamfan If I remember correctly, in my case the partitions were not read after writing the image, but I couldn't really figure out why. Because of that I configured photOS manually as described above.

steamfan, 2021-06-10:
Hello,
with the manual method I was able to write the image to file, manually edit the wpa_suppliant.conf file, boot the Raspberry and access the WebGUI.

Thanks so far for the support

avanc, 2021-12-05:
I'm not able to investigate further, as I don't have MacOS.
Would be great to get additional feedback, if the problem still exist for you and others.

To everyone: Please vote with thumbs down if you faced the same problem, thumbs up if you didn't had problems in combination with MacOS.