avanc/photOS #40: Wiki enhancement: Home Automation - Switch Display on/off from iobroker

escuco, 2021-07-10:
**With this solution you can switch your photOS display on/off with iobroker:**

- create a new datapoint "photoframe" in "Objects" / "0_userdata.0" with type = "datapoint" and datapoint type = "value list"
- edit the properties and change the value field into "String" and predefined Values into "on:on;off:off" and leave the rest at it is
- now create a new blockly script in "Scripts" / "common" and import this script:

```
<xml xmlns="https://developers.google.com/blockly/xml">
  <variables>
    <variable id="8Kkq|_9,fcOPSo`8.$Ha">display_value</variable>
  </variables>
  <block type="on_ext" id="a5zBOnfM{Ty=@::`jLJj" x="38" y="-538">
    <mutation xmlns="http://www.w3.org/1999/xhtml" items="1"></mutation>
    <field name="CONDITION">ne</field>
    <field name="ACK_CONDITION"></field>
    <value name="OID0">
      <shadow type="field_oid" id="w|Y2,9F[V[SZ*^LUvijl">
        <field name="oid">0_userdata.0.photoframe</field>
      </shadow>
    </value>
    <statement name="STATEMENT">
      <block type="procedures_callcustomnoreturn" id="AW|XH|TlW^(3tW{EuNGl">
        <mutation name="switch_display">
          <arg name="display_value"></arg>
        </mutation>
        <value name="ARG0">
          <block type="get_value" id="Q].*sL6LlUX?k,(,06QT">
            <field name="ATTR">val</field>
            <field name="OID">0_userdata.0.photoframe</field>
          </block>
        </value>
      </block>
    </statement>
  </block>
  <block type="procedures_defcustomnoreturn" id="T,=!.@0*gqcXO/Gjbz;i" x="37" y="-287">
    <mutation statements="false">
      <arg name="display_value" varid="8Kkq|_9,fcOPSo`8.$Ha"></arg>
    </mutation>
    <field name="NAME">switch_display</field>
    <field name="SCRIPT">cmVxdWVzdC5wb3N0KHsKICAgIHVybDogICAgICdodHRwOi8vMTkyLjE2OC4yLjE3My9kaXNwbGF5JywKICAgIGZvcm06ICAgIHsgYWN0aW9uOiBkaXNwbGF5X3ZhbHVlIH0KfSwgZnVuY3Rpb24oZXJyb3IsIHJlc3BvbnNlLCBib2R5KSB7CiAgICBpZiAoZXJyb3IpIGxvZyhlcnJvciwgJ2Vycm9yJyk7Cn0pOw==</field>
    <comment pinned="false" h="80" w="160">Display on/off</comment>
  </block>
</xml>
```
[image]

- edit the javascript function "switch_display" and change the ip to your photOS ip.
- start the script

Now you can switch your display state only by changing the value from the datapoint object "0_userdata.0.photoframe" to "on" or "off" from everywhere within iobroker.

If you don't like Blockly and you want to tinker around directly in Javascript, this is the generated code:
```
var display_value;

// Display on/off
async function switch_display(display_value) {
    request.post({
        url:     'http://192.168.2.173/display',
        form:    { action: display_value }
    }, function(error, response, body) {
        if (error) log(error, 'error');
    });
}

on({id: '0_userdata.0.photoframe', change: "ne"}, async function (obj) {
  var value = obj.state.val;
  var oldValue = obj.oldState.val;
  await switch_display(getState("0_userdata.0.photoframe").val);
});
```
For sure there are more elegant solutions, but this is how it works nice for me.

ts617284, 2021-07-16:
Hi, 
sounds very interesting, thanks for the post. I tried it out with my iobroker installation, but unfortunatelly I can't get it to work. :-( your descritpion is very extensive and complete, I don't think I made a mistake there. But switiching of the display doesn't work.  (it works in gui via browser though).
I tried to manually add the script and to directly post the string "off" (instead of reading it from an object), but that doesn't work either. 
Is there a way how I could debug the post request to the website? (and yes, I checked the IP address three times now ;-) ). 
Hier, das ist meine angepasste javascript-Funktion zum Testen:
`console.log('************ Test: Wert display_value in function:   *************');
console.log(display_value);
request.post({
    url:     'http://192.168.1.81/display',
    form:    { action: 'off' }
}, function(error, response, body) {
    if (error) log(error, 'error');
});`

Thx! Tim

avanc, 2021-07-23:
Unfortunately, I'm not  familiar with iobroker. But the js code looks similar to @escuco s example. You may try 

    request.post({ url: 'http://192.168.1.81/display', form: { action: 'off' } }, function(error, response, body) {
      console.error('error:', error); // Print the error if one occurred
      console.log('statusCode:', response && response.statusCode); // Print the response status code if a response was received
      console.log('body:', body); // Print the HTML (might be empty)
    });

to get additional debug output.

@ts617284 Regarding code formatting: empty line before code is mandatory, then indent the code by 4 spaces.

ts617284, 2021-07-25:
@avanc: thank you!
tried your additional output options, but they all are displayed empty. 
```
javascript.0 | 2021-07-25 12:29:36.617 | info | (1052) script.js.photorahmen_skripte.test2: body:
javascript.0 | 2021-07-25 12:29:36.617 | info | (1052) script.js.photorahmen_skripte.test2: statusCode:
javascript.0 | 2021-07-25 12:29:36.617 | error | (1052) script.js.photorahmen_skripte.test2: error:
```
Seems to me I don't get a connection since I don't even receive any repsonse or errore codes... 
hm... stuck at the moment since I don't know how to debug this any further :-(

anyhow, it might be a good idea to put @escuco description of his solution in the wiki at the "Home Automation" section, maybe others have more success with it.

escuco, 2021-07-27:
Sorry for my late answer, @ts617284 @avanc, but I wasn't at home the last days.

I've tested again my own description step by step with a new script and a new data point - just for going sure that I didn't any mistakes. Everything works fine with that.

@ts617284, please note, that if you're using another name than "photoframe" for the data point, you'll have to change the name also in the blockly script at two positions (you'll see that, when you open the blockly script).

You can easily test your script by changing the value from the datapoint from "off" to "on" and from "on" to "off" by using the dropdown menu in the iobroker objects explorer. Confirm the value with the checkmark.

Example screenshot:

[image]

ts617284, 2021-07-31:
@escuco : thank you for your help.  yes I have another name for the datapoint, but I changed it in the script as well. It works correctly, since the javascript sub function gets triggert (I am writing log entries out of the sub function). But my problem is, that the http post request has no effect (and doesn't get response error codes as well). So I think I have a connection error to photOS...

escuco, 2022-04-07:
@avanc , wenn das in die Doku übernommen wird, könnte das Issue geschlossen werden.