avanc/photOS #41: permanent settings after power off?

ts617284, 2021-07-24:
I am not sure, if I understand the concept correctly, but is there a way how my settings survive a power off?

I have had photoOS up and running and connected to my Nextcloud. Everything was working fine.  I shut down the raspberry pi to do some work on the wooden frame and switched power on again several days later. Then I didn't get a network connection any more. I checked the sd card and at least the configured wpa_supplicant.conf file was not on the card any more.

I generated a new one, but I can't seem to get it running again. Every time network connection fails:

```
* Starting wpa_supplicant: failed (brcmfmac)
* Configuring wired network: no device
* panic: rebooting...
```

and if I check the sd card in a card reader, my newly generated wpa_supplicant.conf file is missing again...

this is the file:
```
 update_config=1
 ctrl_interface=/var/run/wpa_supplicant
 network={
   scan_ssid=1
   ssid=6471678431
   psk=**********
 }
```

so, I currently have two questions:
1) do I do something wrong in configuring the network (I had it running some days ago though)?
2) what is the way to have permanent settings which survive a power off?

Thank you for any helps and tips!

Tim

mrlaja, 2021-07-24:
WLAN Signal to week. IT was Always so in my dinning room. Its Always rebooting, Ehen WLAN Signal breaks up

avanc, 2021-07-24:
Regarding wifi configuration: The file on the first partition (boot) is only read once and copied to /data/etc/ (on third partition). So it is totally fine that it is gone after boot: https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf

So it seems you have another problem. Unfortunately, the wifi error messages are never specific. You get the same error message if the signal is weak as mentioned by @mrlaja or if the wifi key is wrong or something else...

ts617284, 2021-07-25:
@avanc @mrlaja 
thank you both for your quick reply and help.
Raspberry was 3 m near wlan router, no walls... so I didn't thought I had a too week signal. Anyways tried it again today and it worked instantly...  still a strange feeling, because I don't know what I did wrong, but okay :-) I take it.
Thank you very much again!
Tim

ts617284, 2021-07-25:
hm... just realized my display is acting up again... 
wlan connectiong error :-( are forcing a reboot.

I think it would be very nice, if the wlan depency is not forcing a reboot when it looses connection (see also #9 ). Instead it should just go on with displaying the already downloaded images (and just showing a error message in the display).

avanc, 2021-08-16:
@ts617284 Were you are able to reproduce your connection problems?

But I agree, that photOS should work also in offline modus. However, I'm not sure of to recover from network problems...

ts617284, 2021-08-19:
@avanc: Thanks for asking! No, not really. Or at the end, I wasn't able to connect at all. I have ordered a new rapsberry, maybe mine had a defect with its wlan module. Or I have to check general wlan networks around here, maybe something is disturbing the lower bands (I usually use 5Ghz bands at home, so habe no reference really).

Maybe an idea for network handling would be the following:
- at reboot trying to connect and if that fails after some retrys display the pictures in buffer (if there are any after reboot? not sure)
- if connection is lost during normal operation, just try to reconnect every minute (or so), otherwise just go on with displaying already buffered pictures (maybe show an error meassage on the display in the corner)
- if I really want to have a reboot and a recovery process of the network, then I just disconnect power and reboot the hard way ;-)

avanc, 2021-11-01:
@ts617284 Any news from your side? If it is not an issue anymore, I would close the ticket.
I track the request to be independent of network in https://github.com/avanc/photOS/issues/9.

ts617284, 2021-11-02:
@avanc : I couldn't find a solution to the unreliable wlan connection. PhotoOS frame is now in a completley different room, where it works much better. So I assume I hade some very strange disturbances, but never figured out what caused it.
But anyways: Yes, please close the ticket. It think it would be important not to completely rely on 100% wlan, but as you said, that is tracked in #9 .

closed 2021-12-01 by avanc