avanc/photOS #42: feature request: portrait mode?

ts617284, 2021-07-25:
Hello,
is there an option to configure a portrait mode? (instead of landscape?). Looked through documentation and didn't find it.
There could be two reasons in my opinion where it could be useful also for others:
- a large junk of your photos are actually portraits of people, then it could be worthwile to orient the frame at the wall upright (long side vertically).
- also maybe some space restrictions on the wall where the frame in portrait just looks better.

I honestly don't know how hard it would be to implement. Probably either directly in the rapsberry display settings or in the section of photOS code, where the zooming/resizing takes place. Also don't know if others really would like to have the option?

Anyhow, big thanks to @avanc for sharing his project and all his work. I have a lot of fun with it!

Tim

avanc, 2021-08-03:
Great idea!
This can be easily done by setting `display_hdmi_rotate=1` in config.txt. (see https://www.raspberrypi.org/documentation/configuration/config-txt/video.md for additional information).

The file is located on the first partition, if you read the SD-Card in your computer. If you are logged into a running photos, the file can be found at /boot/config.txt. You have to remount the partition read-write: `mount -o remount,rw /boot` before editing.

I'll going to add this into the documentation.

avanc, 2021-08-08:
Documented at https://github.com/avanc/photOS/wiki/Screen-Orientation

closed 2021-08-08 by avanc