avanc/photOS #43: Black bars on the display

ostrowsky74, 2021-08-03:
On a TV I have black bars on the left and right for every photo. The photo is cropped.
Have already tried different resolutions, but no change. It seems to be due to the resolution of the TV. Tested a different photOS system, and the same result.

Thank you 

Timo

avanc, 2021-08-03:
Please execute `fbset -s` on the device to see what resolution was detected and if it fits to your TV.

It might be related to overscan? Please try to add `disable_overscan=1` to config.txt. The file is located on the first partition, if you read the SD-Card in your computer. If you are logged into a running photos, the file can be found at /boot/config.txt. You have to remount the partition RW: `mount -o remount,rw /boot` before editing.

Afterwards reboot and see if it helped.

ostrowsky74, 2021-08-03:
I watched with fbset -s, the output is suitable for the TV.

`mode "1920x1280"`
`    geometry 1920 1280 1920 1280 32`
`    timings 0 0 0 0 0 0 0`
`    rgba 8/16,8/8,8/0,8/24`
`endmode`

I took photos so that you could see it better. The images are 1920x1280 in size.
Both are with black bars.

[image]

[image]

Thank you

avanc, 2021-08-03:
To me it seems that the image is not cropped (the red circles hit exactly the end). Thus, my guess is that the image has a different aspect ratio than the screen. 
photOS uses the smartfit option from fbv: https://github.com/godspeed1989/fbv/commit/84aadba6351db7f860df775bef69c214a91cdd14
If scaling the image to fill the whole screen would crop more than x%

avanc, 2021-08-03:
One additional question: After the boot process, is the shown text left aligned to the TV screen or to the image?

ostrowsky74, 2021-08-04:
I looked deeper and played with fbset. The resolution is wrong.
with `fbset -fb / dev / fb0 -g 1280 720 1280 720 16` it fits.
How do I get this value written into `/sys/class/graphics/fb0/modes`?
So that you grab again after a restart?

avanc, 2021-08-04:
So your TV has 720p (1280x720) instead of 1080p (1920x1080)?
I still wonder why it is not detected on bootup. Is the TV connected and on while booting the Raspberry Pi?

The best way to hardcode the resolution is config.txt:
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

I would guess that you should be fine with

    hdmi_group=1
    hdmi_mode=61

But before setting it, you could test what the current working settings are: `tvservice -s`

ostrowsky74, 2021-08-05:
I was watching `tvservice -s` and was a little confused.

> state 0xa [HDMI CEA (95) RGB lim 16:9], 3840x2160 @ 30.00Hz, progressive

Then I tested different resolutions with fstab

> fbset -fb /dev/fb0 -g 2560 1440 2560 1440 32

und

> fbset -fb /dev/fb0 -g 1920 1080 1930 1080 32

and display a full screen.

I then adjusted the config.txt

> gpu_mem=128
> dtparam=i2c_arm=on
> dtparam=i2s=on
> dtparam=spi=on
> dtparam=audio=on
> disable_overscan=o
> dtoverlay=gpio-shutdown,gpio_pin=3, active_low=1,gpio_pull=up
> hdmi_force_hotplug=1
> #hdmi_save=0
> hdmi_group=1
> hdmi_mode=34

And it works now.

Thank you for your help !

avanc, 2021-08-05:
It is an Raspberry Pi 4? Interesting.
I'm going to add this into the documentation.

ostrowsky74, 2021-08-05:
yes, it is the 4GB model

avanc, 2021-08-05:
Added documentation: https://github.com/avanc/photOS/wiki/Screen-Resolution

closed 2021-08-05 by avanc