avanc/photOS #51: Pictures are not scaled to full screen

HirschJunge, 2021-09-25:
I'm using version 0.3.5 and a raspberry pi zero W. When i plug in my laptop to the display the picture is scaled to full screen

HirschJunge, 2021-09-25:
I've tried different things but at the end 
disable_overscan=1 
I've executed tvservice -s and got
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive which correlates to the Displaytype. 
Also fbset -s  promts
mode "1920x1080"

avanc, 2021-10-04:
@HirschJunge Can you provide some photos taken from the screen? I'm not sure what issue you are experiencing.
Regarding quality: fbv (Frame Buffer View) does a very simple scaling ( e.g. no low pass filtering). This results some times in artifacts.

avanc, 2021-11-17:
I'm going to add

    disable_overscan=1

per default.

avanc, 2021-12-05:
<Face Palm> Forgot this change in the last release.

HirschJunge, 2021-12-10:
@avanc unfortunatly the screen is now hanging by my grandparents. I will take some pictures the next time I'm around.

moritzj29, 2022-03-03:
I also see this with the current release, adding `disable_overscan=1` fixes it.

Also the log messages are not in the corner of the screen but shifted inwards so its easily distinguishable from wrong image viewer behaviour/settings.

avanc, 2024-05-14:
Workaround documented here: https://github.com/avanc/photOS/wiki/FAQ#i-still-see-unnecessary-black-bars