avanc/photOS #53: Webdav trust certificate

flomickl, 2021-10-31:
I think an option to trust the certificate from my own Nextcloud address would be create.

At the moment, I get the following message and don't know how to set the certificate to trusted.
Mounting https://XXXXXXX failed: mount.davfs: the server certificate does not match the server name mount.davfs: the server certificate is not trusted.

avanc, 2021-11-01:
Does this helps https://github.com/avanc/photOS/wiki/Self-signed-certificate ?
Or how was the certificate created?

flomickl, 2021-11-06:
Thanks for the reply
I copied the pem file to the location but now the error message has changed to: can't read server certificate /data/photoframe/conf/webdav_cert.pem

`-rw-------    1 root     root        1.7K Nov  6 11:53 webdav_cert.pem`

Any suggestion why?

avanc, 2021-11-08:
Hm, no idea. I rechecked the path and it looks valid: https://github.com/avanc/photOS/blob/master/package/photoframe/etc_photoframe/davfs2_owncert.conf

avanc, 2021-12-01:
You might try manually mounting webdav to see error messages:

    mount.davfs -o ro,conf=/etc/photoframe/davfs2_owncert.conf "YOURWEBDAVURL" /data/photoframe/images_webdav

cs301cs301, 2022-01-12:
you have to chmod the cert file 
`-rw-r--r-- 1 root root 1.7K Nov 6 11:53 webdav_cert.pem `

then it works.

avanc, 2022-01-15:
This problem should be gone with rclone (https://github.com/avanc/photOS/issues/80)