avanc/photOS #54: Support for Raspberry Pi Zero 2 W

avanc, 2021-11-01:
Although the Raspberry Pi Zero W is sufficient to run photOS, the new successor should also be supported.

Have to wait for support in thingOS:  https://github.com/ccrisan/thingos/issues/75

HirschJunge, 2021-12-10:
Does the Zero W image run on the Zero 2 W? Because it's hard to get an Zero W.

avanc, 2021-12-10:
Unfortunately not. The Zero 2 is more similar to Raspberry Pi 3. However, that image won't work neither.

avanc, 2021-12-10:
@HirschJunge I just released a prerelease with RPi Zero2 support:
https://github.com/avanc/photOS/releases/tag/v0.3.7-rc1
You have to flash the RPi3 image. It works on my board without any problems. 

Note: I forgot to update the version number. So after installation it will still say it is v0.3.6 instead of v0.3.7-rc1.

avanc, 2021-12-30:
Now available in release 0.3.7

closed 2021-12-30 by avanc