avanc/photOS #55: Audio OFF/ Ton bei Video ausschalten

Silber-Toaster, 2021-11-11:
Hallo,
wie kann ich den Ton bei einem Video ausschalten?
Der Ton kommt über HDMI Signal in den Monitor. 

Hello,
how can I turn off the sound in a video?
The sound comes into the monitor via HDMI signal. 

Gruß Norman

avanc, 2021-11-11:
You might try

    dtparam=audio=off

in /boot/config.txt

Silber-Toaster, 2021-11-15:
Danke für dein Tip.
Aber leider hat es nicht geholfen.

Thanks for your tip.
But unfortunately it did not help.

avanc, 2021-11-16:
You are right.

    hdmi_ignore_edid_audio=1

did the trick for me (https://www.raspberrypi.com/documentation/computers/config_txt.html#hdmi_ignore_edid_audio).
Let me know if it also works on your side, then I'm going to update the FAQ.

Silber-Toaster, 2021-11-17:
Danke, das war die Lösung, nun funtoniert es. :)

Thanks, that was the solution, now it works. :)

avanc, 2021-12-01:
Documentation updated: https://github.com/avanc/photOS/wiki/FAQ#how-to-deactivate-audio-output-via-hdmi

closed 2021-12-01 by avanc