avanc/photOS #59: Poweroff button support

pmt272, 2021-12-03:
Is there a possibility to add a button and a PIR.
i would like to use the PIR for switching the HDMI on or off and the button to power off the raspberry.

avanc, 2021-12-03:
PIR-Sensor is already supported: https://github.com/avanc/photOS/wiki/Motion-Detection

Power button is not available, but should be possible with a short script, which executes `shutdown`.

wishy370, 2021-12-04:
> Is there a possibility to add a button and a PIR. i would like to use the PIR for switching the HDMI on or off and the button to power off the raspberry.

Add a switch to Pin GPIO3 and Ground
to power on, this code works, power off is not working:
dtoverlay=gpio-shutdown,gpio_pin=3,active_low=1,gpio_pull=up

Whit Raspian, the code for shutdown works also....

pmt272, 2021-12-04:
I know about possibility with PIR and this works very good - sorry, my question was not clear.
The Switch on GPIO3 is working great for me to power on, but the possibility to shutdown would be great to.

wishy370, 2021-12-04:
> The Switch on GPIO3 is working great for me to power on, but the possibility to shutdown would be great to.

same for me...
why is there a difference between raspian and photOS?

avanc, 2021-12-04:
First of all, as the root partition is mounted read-only, a sudden power-off does not harm much. My parents did it for a few months every evening and the SD-Card was never corrupted.

However, I agree that a normal shutdown would be better. I'm going to investigate how _gpio-shutdown_ is working. I would guess that Raspbian is somehow listening to that event and is triggering the shutdown. So it is not only the Kernel involved, but also the OS which is based on Buildroot for photOS and very minimalistic.

avanc, 2021-12-04:
I found it: The solution depends on systemd:
https://github.com/raspberrypi/linux/blob/a1d0f808d4a0b7f7053095cd4ab97a4276bed9ff/arch/arm/boot/dts/overlays/README#L1184

But there is also a hint for other systems...