avanc/photOS #60: Support for openvpn

wishy370, 2021-12-04:
Hi! I don't want to load the photos directly over the internet, but through a VPN tunnel. How can I integrate OVPN in PhotOS?

[](https://www.ovpn.com/de/guides/raspberry-pi-raspbian)

avanc, 2021-12-04:
OpenVPN is available as package in buildroot: https://github.com/buildroot/buildroot/tree/master/package/openvpn

You could give try to build an image with OpenVPN:
* https://github.com/avanc/photOS/wiki/Building-an-Image
* Before step 2 do:
  *  ./build.sh <board_name> menuconfig
  * And select openvpn as additional package.

stonemaan, 2022-02-03:
> OpenVPN is available as package in buildroot: https://github.com/buildroot/buildroot/tree/master/package/openvpn
>
> You could give try to build an image with OpenVPN:
>
> * [...]

Hi, I tried it several times, but it failed building on my Mac and on a pi4 - I cannot find a solution. 
This is, where I find an error:
`sed -i 's,getenv("CCACHE_DIR"),getenv("BR_CACHE_DIR"),' /Users/stonemaan/Desktop/photOS/output/raspberrypi/build/host-ccache-3.7.9/src/ccache.c
sed: 1: "/Users/stonemaan/Deskto ...": bad flag in substitute command: 'O'
make[1]: *** [/Users/stonemaan/Desktop/photOS/output/raspberrypi/build/host-ccache-3.7.9/.stamp_patched] Error 1
make: *** [_all] Error 2`

Is it possible, to install OpenVPN via ssh?
not even apt-get is working..

Thanks

stonemaan, 2022-02-04:
tried again building it on a Pi 4 for a Pi zero - received 
`>>> toolchain-external-custom  Configuring
Cannot execute cross-compiler '/home/pi/Desktop/photOS/output/raspberrypi/host/opt/ext-toolchain/arm-bcm2708/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc'
make: *** [package/pkg-generic.mk:259: /home/pi/Desktop/photOS/output/raspberrypi/build/toolchain-external-custom/.stamp_configured] Fehler 1`
cannot find a solution for that either. do you have any suggestions?

avanc, 2022-02-04:
I don't have experience with MacOS. Maybe this helps:
https://stackoverflow.com/questions/17534840/sed-throws-bad-flag-in-substitute-command/17535174#comment25502270_17535174

This sounds like a solution:
https://stackoverflow.com/a/20827801
But I don't know how to tell buildroot to use gsed instead of sed...

stonemaan, 2022-02-04:
unfortunately no.. but I gave up doing it on Mac. 
try to do it on a Pi 4 with Raspbian.. but I unfortunately cannot find a solution for the error mentioned above.. 
`>>> toolchain-external-custom Configuring Cannot execute cross-compiler '/home/pi/Desktop/photOS/output/raspberrypi/host/opt/ext-toolchain/arm-bcm2708/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc' make: *** [package/pkg-generic.mk:259: /home/pi/Desktop/photOS/output/raspberrypi/build/toolchain-external-custom/.stamp_configured] Fehler 1`