avanc/photOS #61: Removal of pictures when webdav server is offline

b3nsk0, 2021-12-05:
Hi,
it seems that if the webdav server is offline, all local pictures were removed. Is it possible to change this behavior so that pictures were only synchronized (removed/changed) when the webdav server is online?

avanc, 2021-12-05:
Hm, photOS should check if webdav is mounted before initiating the sync:
https://github.com/avanc/photOS/blob/66e42b553b47b78b6254827ae251c1df50ace44f/package/photoframe/photoframe.sh#L85-L95

So I'm not quite sure what happens in your case. Can you give me some additional information to investigate further? Was the server not reachable or down?

b3nsk0, 2021-12-05:
of course. I'm hosting the server on my synology nas and when the server is down, all synced pictures were removed, when I turn the server on the synchronization starts and after downloading everything is normal again.

avanc, 2021-12-09:
Thanks, I could reproduce the problem. I'm still investigating...

avanc, 2021-12-09:
mount.davfs: connection timed out two times;
    trying one last time
    mount.davfs: server temporarily unreachable;
    mounting anyway

Why the hell is davfs "mounting anyway" if the server is unreachable?

cs301cs301, 2022-01-07:
What's about rclone, maybe a better way?

michkrom, 2022-07-08:
Possibly add check of the mount location for any files?

avanc, 2022-07-18:
This  would mean that you cannot wipe all images remotely. But I agree the impact is smaller than the current situation.