avanc/photOS #65: Feature Request - Slideshow Transitions

supertevran, 2021-12-10:
Would'nt it be great to introduce a slideshow effect like seen on radioparadise.com? The pictures are constantly moving a little bit and there is a kind of transition effect of one to the next picture included as well. Just check it out. http://www.radioparadise.com

universalappfactory, 2021-12-17:
I'm not sure if this is possible with fbv which shows the pictures.
But I played a bit with https://www.raylib.com/ maybe this an appropriate  library

avanc, 2021-12-30:
It's a challenge. Until now I avoided hardware acceleration like OpenGL to keep it simple (and I don't have any experience ;-) ).

cs301cs301, 2022-01-12:
Couldn't `fbi` be the solution here? It supports blend. [http://manpages.ubuntu.com/manpages/bionic/man1/fbi.1.html](http://manpages.ubuntu.com/manpages/bionic/man1/fbi.1.html)