avanc/photOS #67: manual buttons not working

chrgachrga, 2021-12-14:
Hi,
I was checking on the new feature for physical buttons. My conf looks as follows:
GPIO_PIN_NEXT=5
GPIO_PIN_PREVIOUS=6
GPIO_PIN_PLAY=0
GPIO_ACTION_VALUE=0
SLIDESHOW_DELAY=10

If I now add a button between 3.3v and GPIO 5 or 6, nothing happens.
I am on a raspberry pi 2.

Can you confirm that the config and button is right? Or am I missing anything?

thanks,
chrga

universalappfactory, 2021-12-17:
Hi,

I have prettry the same config and it works for me with a rasperry pi zero. Have you set GPIO_ACTION_VALUE according to your setup (pullup or pulldown resistor)?

And you may check this output in your console:
cat /sys/class/gpio/gpio6/value

there you should see if the button is pressed or not (press the button and run the command)

universalappfactory, 2021-12-17:
Ah and I think there is an error in the is_gpio_pressed function.

if [ "$(cat /sys/class/gpio/gpio${GPIO_PIN_NEXT}/value)" -eq "${GPIO_ACTION_VALUE}" ] 

should be

if [ "$(cat /sys/class/gpio/gpio${1}/value)" -eq "${GPIO_ACTION_VALUE}" ]

I think.

avanc, 2021-12-17:
@universalappfactory Thanks for pointing that out.

avanc, 2021-12-27:
@chrgachrga I fixed the issue emntioned by @universalappfactory. However, I#M not sure if this is your issue, as _something_ should happen also with the bug. Please reopen with additional information if needed.

commit 3fef2e26 (2021-12-27): Fix GPIO issue

Fixes #67
closed 2021-12-27 by avanc