avanc/photOS #68: WLAN USB Adapter

JSAnyone, 2021-12-17:
Hi,
Are WLAN USB Adapters like the famous Edimax supported?
They are easy to use on Debian, but it looks like it isn't that easy on buildroot:
https://www.grappendorf.net/tutorials/nerves-pizero-edimax.html

JSAnyone, 2021-12-19:
I tried some things out and realized that it is already possible without compiling a new image:

1. Flash the image and add wifi credentials like usually
2. Boot your Raspberry Pi with a ethernet connection
2. Start a SSH connection to the Raspberry Pi
3. Log in with user `root` and empty password
4. Remount filesystem: Type `mount -o rw,remount /`
5. Open modules file: Type `vi /etc/modules`
6. Edit modules file: Type `i` to start insert mode, type `R8188eu`, press `esc` to leave insert mode, type `:wq` to save and close the file
7. Type `reboot` to restart photOS

That's all it took to get my Edimax WLAN Adapter working on a Raspberry Pi. 
If you think it might be interesting for others I can add it to the Wiki of this repository.

avanc, 2021-12-30:
Reminder to myself to add this to the wiki. I didn't knew it as well :-)

ueen, 2024-12-24:
> 7\. type `R8188eu`, press `esc` to leave insert mode,

@jscmidt please clarify, did you override something with this or should one add it in a new line? 
Merry Christmas 🎄

jscmidt, 2024-12-24:
If I remember it correctly the file was empty, but you shouldn’t do anything wrong with just appending.

ueen, 2024-12-24:
Awesome thank you for the quick reply, I appended but it didn't work unfortunately, I have the TP-Link TL-WN725N and raspberry pi 1, it usually should be plug n play...

jscmidt, 2024-12-25:
Unfortunately it isn’t plug and play, since as far as I know photOS is build on build root and not raspberry Pi OS, so drivers might be different…

ueen, 2024-12-25:
True, we figured it out so far that we came set iwconfig via ash manually and get a connection. 
But we can't persist it after reboot, the config file is not working or documented to little for us