avanc/photOS #69: Handle root access

avanc, 2021-12-30:
Currently, the device is not secured with a password.

I see two options to handle this:
1. Create initial root password and first boot and show it in boot screen.
   * Secure against purely remote attacks, but still no security if the attacker gets access to the device.
2. Deactivate ssh completely and add option in WebUI to activate it.
  * In this case, the WebUI should also be password protected.

Ideas and suggestions are welcome to make photOS secure by default while still allowing for manual customization.

cs301cs301, 2022-01-12:
Option 2 seems to be the best. And in the case that you cannot access the device via SSH connection due to a non-public IP (cable internet), add the functionality of remote_settings.txt via the cloud. ;-)

MisterMike, 2022-04-19:
I'd vote for option 2, too. I don't know if it would be hard to make the WebUI https ? I'd also vote for a WebUI Setting to customize the frame's (local) URL.