avanc/photOS #74: Better use Pi Zero 1 w than Pi Zero 2 w - recommendation

cs301cs301, 2022-01-12:
If you have the choice of using a Pi Zero 1 W or Pi Zero 2 W, go for the Pi Zero 1 W.
The problem is the wifi. The Pi Zero 2 generates an interference signal when output via the HDMI, which causes the WiFi to collapse.
This occurs from FullHD resolution 60 Hz. The Pi Zero 1 also has the problem, but the wifi still works with FullHD.

[https://elinux.org/RPi_Configuration](url)
I set resolution in the `config.txt `
```
hdmi_drive=1
hdmi_group=2
hdmi_mode=82
```
is possible to / works on Pi Zero 1 W
```
hdmi_drive=1
hdmi_group=1
hdmi_mode=16
```

On Pi Zero 2 W, the Wi-Fi dies and a kernel panic reboots the Pi. You can proof this by pinging the Pi.
With Google you can find many reports of this HDMI/Wi-Fi problem.

I will try with an extra OTG cable  and Wi-Fi dongle. Maybe the distance help.

Monitor Fujitsu B24W-7, Pi Zero 1 / 2 w, FLIRC Case.

cs301cs301, 2022-01-15:
When testing with a WiFi dongle, I noticed that the country settings from wpa_supplicant.conf were not adopted. According to this, channel 1 was used, which is mentioned for the problems with HDMI and dying WLAN. I had to set it with ` iw set reg DE`. Now channel 13 is corectly set. The internal WLAN hasn't died since then either.

avanc, 2022-01-16:
Could you paste links of the other discussions as reference. I want to put this hint into the Wiki.

cs301cs301, 2022-01-16:
[https://forums.raspberrypi.com/viewtopic.php?t=247982&sid=55d12dc87af911467cecabd76a58e3ac&start=25#p1585816](https://forums.raspberrypi.com/viewtopic.php?t=247982&sid=55d12dc87af911467cecabd76a58e3ac&start=25#p1585816)

[https://forums.raspberrypi.com/viewtopic.php?t=247982&sid=55d12dc87af911467cecabd76a58e3ac&start=25#p1560446](https://forums.raspberrypi.com/viewtopic.php?t=247982&sid=55d12dc87af911467cecabd76a58e3ac&start=25#p1560446)

[Channel 1 ist mentioned in the link of post 1, too.](https://hackaday.com/2019/11/28/raspberry-pi-4-hdmi-is-jamming-its-own-wifi/) 

...but the main solution is to use 5GHz instead of 2.4GHz!