avanc/photOS #78: Downloads from server, the whole day - Problem

cs301cs301, 2022-01-13:
I have a problem with webdav. It seems, that the images are loaded from the server every minute every day. Although the files are available locally. 

shortened webdav protocoll of my Synolgy NAS, IP is anonymized..

```
WebDAV,2022/01/13   18:11:35,8.8.8.8,photoframe,download,Datei,5.00   MB,/photoframe/IMG_20190317_125010.jpg
WebDAV,2022/01/13   18:10:21,8.8.8.8,photoframe,download,Datei,2.56   MB,/photoframe/IMG_20180207_103914.jpg
WebDAV,2022/01/13   18:09:23,8.8.8.8,photoframe,download,Datei,2.07   MB,/photoframe/IMG_20190315_221541.jpg
WebDAV,2022/01/13   18:07:22,8.8.8.8,photoframe,download,Datei,2.49   MB,/photoframe/IMG_20180524_111335.jpg
WebDAV,2022/01/13   18:05:45,8.8.8.8,photoframe,download,Datei,2.25   MB,/photoframe/IMG_20180609_120357.jpg
WebDAV,2022/01/13   18:04:59,8.8.8.8,photoframe,download,Datei,2.28   MB,/photoframe/IMG_20190309_121815.jpg
WebDAV,2022/01/13   18:03:17,8.8.8.8,photoframe,download,Datei,2.71   MB,/photoframe/IMG_20190116_110835.jpg
WebDAV,2022/01/13   18:01:58,8.8.8.8,photoframe,download,Datei,2.49   MB,/photoframe/IMG_20181024_093631.jpg
WebDAV,2022/01/13   17:59:54,8.8.8.8,photoframe,download,Datei,2.27   MB,/photoframe/IMG_20190309_121140.jpg
WebDAV,2022/01/13   17:59:07,8.8.8.8,photoframe,download,Datei,2.10   MB,/photoframe/IMG_20180904_194147.jpg
WebDAV,2022/01/13   17:58:14,8.8.8.8,photoframe,download,Datei,2.30   MB,/photoframe/IMG_20190309_121811.jpg
WebDAV,2022/01/13   17:57:48,8.8.8.8,photoframe,download,Datei,2.38   MB,/photoframe/IMG_20180715_211143.jpg
...
...
WebDAV,2022/01/12   22:56:27,8.8.8.8,photoframe,download,Datei,5.07   MB,/photoframe/IMG_20190612_183614.jpg
WebDAV,2022/01/12   22:56:18,8.8.8.8,photoframe,download,Datei,4.47   MB,/photoframe/IMG_20190612_183241.jpg
WebDAV,2022/01/12   22:56:10,8.8.8.8,photoframe,download,Datei,4.15   MB,/photoframe/IMG_20190612_183122.jpg
WebDAV,2022/01/12   22:56:00,8.8.8.8,photoframe,download,Datei,4.77   MB,/photoframe/IMG_20190612_183033.jpg
WebDAV,2022/01/12   22:55:56,8.8.8.8,photoframe,download,Datei,2.25   MB,/photoframe/IMG_20190609_173354.jpg
WebDAV,2022/01/12   22:55:39,8.8.8.8,photoframe,download,Datei,7.19   MB,/photoframe/IMG_20190607_211531.jpg
WebDAV,2022/01/12   22:55:28,8.8.8.8,photoframe,download,Datei,5.86   MB,/photoframe/IMG_20190602_184032.jpg
WebDAV,2022/01/12   22:55:19,8.8.8.8,photoframe,download,Datei,4.65   MB,/photoframe/IMG_20190531_202201.jpg
WebDAV,2022/01/12   22:55:08,8.8.8.8,photoframe,download,Datei,5.69   MB,/photoframe/IMG_20190531_202141.jpg
WebDAV,2022/01/12   22:55:03,8.8.8.8,photoframe,download,Datei,3.05   MB,/photoframe/IMG_20190530_203144.jpg
WebDAV,2022/01/12   22:54:52,8.8.8.8,photoframe,download,Datei,5.69   MB,/photoframe/IMG_20190530_182121.jpg
WebDAV,2022/01/12   22:54:40,8.8.8.8,photoframe,download,Datei,5.16   MB,/photoframe/IMG_20190530_182109.jpg
WebDAV,2022/01/12   22:54:32,8.8.8.8,photoframe,download,Datei,4.07   MB,/photoframe/IMG_20190530_182105.jpg
WebDAV,2022/01/12   22:54:27,8.8.8.8,photoframe,download,Datei,2.51   MB,/photoframe/IMG_20190530_175318.jpg
WebDAV,2022/01/12   22:54:10,8.8.8.8,photoframe,download,Datei,8.41   MB,/photoframe/IMG_20190530_173054.jpg
WebDAV,2022/01/12   22:53:56,8.8.8.8,photoframe,download,Datei,7.11   MB,/photoframe/IMG_20190530_173041.jpg
WebDAV,2022/01/12   22:53:47,8.8.8.8,photoframe,download,Datei,4.87   MB,/photoframe/IMG_20190530_172620.jpg
WebDAV,2022/01/12   22:53:36,8.8.8.8,photoframe,download,Datei,4.93   MB,/photoframe/IMG_20190530_172613.jpg
WebDAV,2022/01/12   22:53:30,8.8.8.8,photoframe,download,Datei,2.69   MB,/photoframe/IMG_20190530_172525.jpg
WebDAV,2022/01/12   22:53:26,8.8.8.8,photoframe,download,Datei,2.46   MB,/photoframe/IMG_20190530_172520.jpg
WebDAV,2022/01/12   22:53:18,8.8.8.8,photoframe,download,Datei,3.83   MB,/photoframe/IMG_20190530_172154.jpg
WebDAV,2022/01/12   22:53:05,8.8.8.8,photoframe,download,Datei,6.94   MB,/photoframe/IMG_20190503_193702.jpg
WebDAV,2022/01/12   22:53:01,8.8.8.8,photoframe,download,Datei,1.94   MB,/photoframe/IMG_20190429_062325.jpg
WebDAV,2022/01/12   22:00:13,8.8.8.8,photoframe,download,Datei,80.28   KB,/photoframe/1b8d0ce463284d61b1c2d139fe1114c2.jpg
WebDAV,2022/01/12   22:00:13,8.8.8.8,photoframe,download,Datei,56.68   KB,/photoframe/11-04-04_1653.jpg
WebDAV,2022/01/12   22:00:13,8.8.8.8,photoframe,download,Datei,40.73   KB,/photoframe/11-04-04_1651.jpg
WebDAV,2022/01/12   22:00:11,8.8.8.8,photoframe,download,Datei,955.68 KB,/photoframe/100_0091.jpg
WebDAV,2022/01/12   22:00:09,8.8.8.8,photoframe,download,Datei,1.24 MB,/photoframe/1.jpg
WebDAV,2022/01/12   22:00:03,8.8.8.8,photoframe,download,Datei,3.23   MB,/photoframe/06072013337.jpg
WebDAV,2022/01/12   22:00:03,8.8.8.8,photoframe,download,Datei,66.97   KB,/photoframe/05-05-04_1853.jpg
...
...
WebDAV,2022/01/12   10:02:54,8.8.8.8,photoframe,download,Datei,2.78   MB,/photoframe/IMG_20180324_155022.jpg
WebDAV,2022/01/12   10:01:15,8.8.8.8,photoframe,download,Datei,2.78   MB,/photoframe/IMG_20180324_155022.jpg
WebDAV,2022/01/12   10:00:36,8.8.8.8,photoframe,download,Datei,2.78   MB,/photoframe/IMG_20180324_155022.jpg
```

There is an error message displayed too:
`Monting https://x.x.x.x:5006/photoframe/ failed: mount.davfs: http://x.x.x.x:5006/photoframe is alrady mounted on /data/photoframe/images_webdav`

Can the problem be the sync cron job? This is started every 30 minutes, but the download of all files takes well over 1 hour.

avanc, 2022-01-13:
While diving into rclone, I noticed that webdav normally does not provide mtime (time of last modification): https://rclone.org/webdav/#modified-time-and-hashes
Although, Nextcloud does support it, it might be not supported by davfs2. Thus, rsync might fall back to checksum comparison of files, where it has to read the whole file just to see that remote and local don't differ.

avanc, 2022-01-13:
Hm, my initial thoughts might be wrong. The checksum thing has to be acivated manually:
https://en.wikipedia.org/wiki/Rsync#Determining_which_files_to_send

In addition, I checked how long the sync takes on my side if no changes are available (1.7GB images and videos in total):  less than 4 seconds

cs301cs301, 2022-01-14:
I have cleared the log and this is the time for a complete sync / checksum run.
13.01.22 20:13:17 first picture to
14.01.22 00:07:59 last picture

1066 files, 2.46 GB
No idea why it takes so long.

cs301cs301, 2022-01-14:
I have to quit here. Display is back at home and the problem is a sum of dying wifi, panic reboot and sync time. Zero 1 dies with FHD resolution too. Have to wait for the wifi dongle.
Pls close or remove this issue.

cs301cs301, 2022-01-14:
I do parallel testing with different rsync params, `rsync -vt --size-only` at moment, and it works fine.

In addition, I noticed that the NTP service was not activated on the NAS. On the Pi it is activated. Maybe that's the reason for the endless sync.

cs301cs301, 2022-01-14:
> Hm, my initial thoughts is probably wrong. The checksum thing has to be activated manually: https://en.wikipedia.org/wiki/Rsync#Determining_which_files_to_send
> 
> In addition, I checked how long the sync takes on my side if no changes are available (1.7GB images and videos in total): less than 4 seconds

Only 4 seconds? Are your data stored on a public cloud or on a private server? If so, what's your upload speed?

avanc, 2022-01-14:
> Maybe i misunterstood this `--checksum`, i need it to transfer less data, but the calculating time ist higher.

Checksum is used to really make sure, that the files are the same. Think of the case where a file was changed, but the size is still the same and the modify time was not changed.
The problem with checksum is, that rsync calculates the checksum on the device. To calculate the checksum, the file has to be transferred completely.

avanc, 2022-01-14:
> Only 4 seconds? Are your data stored on a public cloud or on a private server? If so, what's your upload speed? Or is that the sync time with server in LAN?

Nextcloud on a remote virtual server and photoframe connected via WiFi. But as said, no new files on the server.
Just comparing mtime and size should not take much time.
`--size-only` sound like a good idea if your webdav server does not support mtime.

cs301cs301, 2022-01-19:
Pi Zero 2W is now connected via an USB/LAN adapter and rsync works with "-vrm --size-only". So I'm closing this topic.

closed 2022-01-19 by cs301cs301