avanc/photOS #84: Use open WiFi network

timmmerlan, 2022-04-05:
Hello,
we tried to set up photOS on a RasPi 4 using an open WiFi-Network. We tried leaving the "psk" parameter empty as well as completely removing it, but the Pi didn't connect to the WiFi. What configuration should we use?
Thank you in advance for your help!
Best regards,
Tim

escuco, 2022-04-05:
Hi Tim,

for free wifi networks this settings in wpa_supplicant.conf should work (not tested):

```
network={
ssid="your_open_wifi_ssid"
key_mgmt=NONE
}
```

You'll have to edit the wpa_supplicant.conf by your own, please have a look to point 4.ii from this installation guide:
[https://github.com/avanc/photOS/wiki/Installation](https://github.com/avanc/photOS/wiki/Installation)

Please make sure, that your open (guest) wifi has no extra front page, that has to be confirmed.

@avanc Maybe this can be added to the "wifi" wiki page (https://github.com/avanc/photOS/wiki/Wifi)

timmmerlan, 2022-04-05:
Thank you! It seems to connect - somehow. But I get the error (on the TV) "_Mounting /my/directory/ failed: mount.davfs: the server certificate is not yet valid_"
Do you have an idea how I can fix that?

timmmerlan, 2022-04-05:
Thank you! It seems to connect - at least I don't get an error on the TV. But I cannot connect from the browser nor from SSH because it cannot find the page. Do you have an idea what that might come from?

timmmerlan, 2022-04-06:
I don't yet have access to the router so I could check, but I am 99% sure I am in the same network - the RasPi wouldn't start without a network connection (it is the only one in the config file) and there is no similar named network the PC could be in

escuco, 2022-04-07:
Maybe @avanc has an idea 🤔