avanc/photOS #93: Slideshow Delay WebUI problem

faixk, 2022-10-26:
Hi, 
at first, great work.
I use version 0.3.7 and want to set the slideshow delay, but there are no changes and the pictures change in less than 1 second.
It doesn't matter if I put the delaytime on 5 or 20 seconds, nothing changes.
What am I doing wrong? The rest in the WebUI works well.
--
In my idea of usage I will also just show one picture a day, which I send via email to my webdav. Works perfect.
So I would just put the delaytime on a maximum level, so the picture will not "refresh" so often.
is there a maximum?

faixk

Partyarti777, 2024-12-15:
Hey,

This helped me out, I went into nano **/usr/bin/photoframe.sh** and commented out these lines:

```
#SLIDESHOW_DELAY=10
#SHUFFLE=false
#SHOW_FILENAME=false
#SHOW_VIDEOS=true
#SMARTFIT=30
```

After that, I was able to adjust the settings through the web UI.

avanc, 2024-12-18:
@faixk There is no real limit in the delay.
https://github.com/avanc/photOS/blob/master/package/photoframe/photoframe.sh#L246C8-L246C69
Even weeks should work :-)

@Partyarti777 Did you changed the values before? The DELAY should be 3 by default:
https://github.com/avanc/photOS/blob/master/package/photoframe/photoframe.sh#L22

I wonder why they are not overwritten by https://github.com/avanc/photOS/blob/master/package/photoframe/photoframe.sh#L37

@faixk Does the file /data/photoframe/conf/photoframe.conf exists on your device?