avanc/photOS #95: Space key in Webdav URL

gklank, 2022-11-29:
Hi,

great stuff, impressive!!!

But, my family members are no IT guys, just using it as end user.

Therefore the picture- and video-names includes at least "space key/letter" in foldername and/or picture-name.
German "Umlaute" would be the second wish/question...

Is there any possibility how to get "space key/letter" into the Webdav URL?

Regards

Gerhard

foxbox001, 2022-12-24:
Try replacing the space with %20 in the WebDAV URL box

example if your folder is called "My Pictures" try entering it My%20Pictures. %20 indicates a space.

https://www.w3schools.com/tags/ref_urlencode.ASP

toonemastic, 2025-01-02:
Here is the solution in 2 steps.

1) In file /usr/bin/photoframe.sh :
Change the line 64 from 
    array=($firstline)
into
    eval "array=($firstline)"

2) And put " (quotes) around the WebDAV URL in the webgui