avanc/photOS #101: Feature Request - Show pictures without synching

gerdf, 2023-04-13:
Hi there, I have a NAS that is always running so there's no need to sync files. Another problem with the syncing is, that the space on the Raspberry PI's SD-Card is too small. I've got ~260GB of pictures that is growing and growing. So it would be good to disable the sync process and use the source directly.

avanc, 2023-04-13:
Sounds good. But development slowed down in the past.

gerdf, 2025-02-16:
I modded the scripts for my own needs so that the pictures are directly read from an SMB share without syncing and so on.
But I removed all that WebDAV stuff so it's not compatible to the original branch anymore. What s the best way to share my files?

avanc, 2025-02-17:
@gerdf Can you share your changed files in this ticket? Or how many you had to change?
Might help others as inspiration and me also for the future.

gerdf, 2025-02-18:
I'm just migrating from a Pi3 to a Pi4 therefore I've to install the photOS Pi4 image and I'm documenting my doings.
When I've finished I'll post it here.

schmidttrick, 2026-08-30:
I’m also interested in running PhotOS without syncing, because I’m also running a NAS.

Can anyone give me a hint which files I have to change?