avanc/photOS #104: Offline Mode: Kernel Panic, when no known SSID is present

kumzugloom, 2023-05-23:
The readme quotes: 

> **Offline**: Works also if no internet connection is available.

However, if no already known SSID is present, my device is forced into eternal reboots due to kernel panics. Is there any documentation on how to achieve the offline mode?

kumzugloom, 2023-05-28:
In case anyone else wonders:
According to [this ](https://github.com/ccrisan/thingos/wiki/netwatch.conf) you can deactivate automatic reboot on unavailable WiFi Connection by changing /etc/watch.conf
from 

> LINK_WATCH="true"

 to 

> LINK_WATCH="false"

closed 2023-05-28 by kumzugloom

avanc, 2023-05-30:
@kumzugloom Sorry that I had no time to respond. Thanks for the details, I added it into the documentation: https://github.com/avanc/photOS/wiki/FAQ#how-to-avoid-reboots-if-network-connection-is-not-working