avanc/photOS #105: question: Using photOS offline

Notonegoro, 2023-07-05:
Hi, 
is there an option, that i could run photOS offline, because on the location i want to use it, there is no wifi. 
As a workaround i could think of trying to run the WebDav server local on the same raspberry. (I guess that's to experimental)
So is it maybe possible to just use a folder on the SD-card or on a mounted external storrage?

jinks113, 2023-07-08:
My workaround was to use an old router (TPlink WR710) after setup (connected to the internet). Then the pictures are downloaded. This may also be done manually to this folder: /data/photoframe/images_local/
photOS receives an IP address from the router and is satisfied to start.

Notonegoro, 2023-07-08:
That sounds like a solution, but unfortunately i would prefer to run it without a additional device (since i would have to purchase it). 
Would i be possible to add a flag in the config for offline use?

Surfer2010, 2023-09-04:
Would be a better solution in my eyes to start up no matter if there is a wifi signal or not, show the saved pictures and check (using a cronjob) ever x min/hours of a signal is available. 
Once connected trigger the sync process. 

What do you think?
How could this be done

Mathes75, 2024-08-21:
I stumbled upon the same issue and in the WiKi I found a hint that helped me.

My scenario: Pi Zero W boots up using my mobile phone hotspot to sync and to prevent Panic reboots. Shutting down the hotspot or moving away from Pi keeps it alive, cycling the stored pictures.
For updating, I need to cut power and reboot with my hotspot available.

Solutions for your request @Notonegoro:

Create file os.conf by using SSH, enabling PhotOS to boot without network:
`> ssh root@photos-xxxxxxxx
> echo "OS_NETWORKLESS=true" >> /data/etc/os.conf
> sync
> reboot`
In that case, PhotOS will boot without network, no update, no remote control.

To use same setup as described above, please use following SSH commands:
`> ssh root@photos-xxxxxxxx
> echo "LINK_WATCH=false" >> /data/etc/watch.conf
> sync
> reboot`

The instructions given at the end of the [FAQ](https://github.com/avanc/photOS/wiki/FAQ#how-to-avoid-reboots-if-network-connection-is-not-working) are not working as they are creating a file netwatch.conf that is not used by OS.