avanc/photOS #113: How to support the project?

michaelsmolka, 2024-01-06:
Hi,

recently I came up to this project and now have it running on three instances. 
Although there are some issues and I have a few ideas on hoe wot improve it, for now I just want to say "thank you".

Is there a way to support the project as a "non-coder"? Maybe a "buy-me-a-coffee"-link or so?

avanc, 2024-01-11:
Thanks for the offer. As I have currently not much time to improve photOS I would feel uncomfortable to get something.

lanzer, 2024-05-14:
Hey just add a "buy me a coffee link".  Even old code if used today still deserve some love.  This is by far the best solution out there.  The alternative is to spend $50 on a 2GB Raspberry Pi just so you can have a picture frame in portrait mode.  It's nuts.