avanc/photOS #114: Display is switched off every hour (V 0.3.7)

TuxMaus, 2024-01-10:
The display is switched off every full hour. It seams as there is a corn job executed every hour but I can`t find any suitable cronjob.

According to `/data/etc/crontabs/root`, the display should be switched off at midnight but not every hour:
```
0 8 * * * /usr/bin/photoframe.sh display on
0 24 * * * /usr/bin/photoframe.sh display off
```

And also `/etc/crontabs/root` has no corresponding entry:
```
13 * * * * /usr/sbin/logrotate /etc/logrotate.conf
0 2 * * 0 /usr/sbin/https-update
*/15 * * * * /usr/sbin/dyndns-update
*/30 * * * * /usr/bin/photoframe.sh sync
```
Any ideas where else to search?

avanc, 2024-01-11:
Hm, everything looks OK. Can you also check that /etc/cron.d is empty and /data/etc/cron.d does not exist?
When its turned off, can you activate it again via web interface?

TuxMaus, 2024-01-11:
Sorry, I forgot this info. 

Yes, /etc/cron.d is empty and /data/etc/cron.d does not exist. The Display can be activated via the web interface and also via `vcgencmd display_power 1`. 

But every time I swith the display on it will be switched off at full hour:
```
11.01.2024 06:56: display_power=1
11.01.2024 06:57: display_power=1
11.01.2024 06:58: display_power=1
11.01.2024 06:59: display_power=1
11.01.2024 07:00: display_power=0
11.01.2024 07:01: display_power=0
11.01.2024 07:02: display_power=0
```
I have absolutely no idea why the display is switched off. Is there any service running which can switch the display off?

TuxMaus, 2024-01-12:
Issue solved! 

It was just a silly mistake in the time code for cron. The time code for the hour is defined in the range from 0 to 23. `24` for midnight is not known by cron and - as it seems - interpreted like the `*`. Therefore, the script was executed every hour and not just at midnight. 

A small hint in the WebUI could problaby prevent others from doing the same mistake.