avanc/photOS #123: error: the server does not support locks mount.davfs

Schimy, 2024-06-06:
Hi, I have installed photOS on an RPi 2 and created a new user on my Nextcloud and shared a directory for the photos. 
When integrating via WebDAV, I now get the following error message:

`Mounting https://<url-to-my-nextcloud>/remote.php/dav/files/<user>/<share> failed: mount.davfs: the server does not support locks mount.davfs: Mounting failed. 404 Not Found
`

Can anyone here help me and tell me what I am doing wrong?

many thx and so long,
schimy

ssanikovich, 2024-06-14:
The same issue

avanc, 2024-06-19:
Hm, I searched a little bit for this error message, could not find much.
There seems to be an app available: https://github.com/nextcloud/server/issues/1308
But I didn't installed any additional apps on my instance and it is working as expected...

In addition, there shouldn't be a lock, as the files are only copied from source, but not modified...
https://wiki.archlinux.org/title/Davfs2#Creating/copying_files_not_possible_and/or_freezes

Schimy, 2024-06-23:
had experimented a bit with fbi and davfs in the meantime. ... also a good project here: https://www.instructables.com/DIY-Raspberry-Pi-24-Digital-Picture-Photo-Frame/

then a fresh installation of photoos ... and now, everything runs out-of-the-box ... magic!
thx for the support

closed 2024-06-23 by Schimy