avanc/photOS #124: No images found

BigChris, 2024-06-14:
I have reinstalled photOS and would like to write the images manually to the SD card.
However, I always get the message "No images found"
Where do the images have to be stored so that photOS can find them?

avanc, 2024-06-19:
Is the webdav endpoint configured in the [WebUI](https://github.com/avanc/photOS/wiki/Configuration)?
First sync from e.g. [Nextcloud](https://github.com/avanc/photOS/wiki/Nextcloud) may take some time.

BigChris, 2024-06-19:
Thank you very much for your message.
I would like to use photOS without WebDAV. Upload the images via (s)FTP, for example. If you don't have a webdav server, the effort might be too high just to set one up.

To do this, I would like to move the images to the local_images folder and initiate the synchronisation process manually. But I don't know how to do that. 
In principle, you can also write to the text file manually, but that would be more work. 

I love the photOS project very much and would be delighted if it could be actively maintained again.
I would also be very happy to support it as much as I can.

avanc, 2024-06-19:
You should store your images under /data/photoframe/images_local.
Afterwards, you have to create the file list similar to
https://github.com/avanc/photOS/blob/master/package/photoframe/photoframe.sh#L102

`find /data/photoframe/images_local -type f -iname '*\.jpg' -o -iname '*\.jpeg' -o -iname '*\.png' -o -iname '*\.mp4' -o -iname '*\.mov' | sort > /data/photoframe/conf/filelist.txt`

closed 2025-02-17 by avanc