avanc/photOS #127: Home Assistant / RESTful: Display on/off via HTTP request

tiphie21, 2024-10-03:
Hi,

I just tried to integrate this great software photOS (thanks for that!) into an automation in Home Assistant.
So, I want to turn off the display when nobody is at home and turn in on when someone is at home and there is more power out of solar energy than I can use currently.
But unfortunately I could not get it working ... even with the FAQ and other topics using NodeRED or ioBroker.

What I tried as far is following:

1) In configuration.yaml I created a new RESTful part with `rest_command`:
```
rest_command:
  rest_photos_display:
    url: "http://192.168.xx.xx/display"
    method: POST
    payload: '{ action={{ wert_action }} }'
```

2. And in Development tools I tried the action `rest_command`:
```
action: rest_command.rest_photos_display
data:
  wert_action: "off"
```

But after I ran this request I got a HTTP-400 ...
```
content: "<html><title>400: Bad Request</title><body>400: Bad Request</body></html>"
status: 400
```

Has anyone a clue that could help me?
Thanks alot in advance :)

tiphie21, 2024-10-03:
Okay, I did not expect it but ChatGPT helped me out very well.

So, here is the solution of ChatGPT which worked for me:

1. configuration.yaml: insert this snippet
```
rest_command:
  set_photos_display:
    url: "http://<IP-address>/display"
    method: "POST"
    content_type: "application/x-www-form-urlencoded"
    payload: "action={{ action }}"
```

2. action in the automation:
```
action:
      - service: rest_command.set_photos_display
        data:
          action: "on"  # activate
```
or for deactivate:
```
action:
      - service: rest_command.set_photos_display
        data:
          action: "off"  # deactivate
```

closed 2024-10-03 by tiphie21

escuco, 2024-10-03:
Maybe @avanc can put this into the wiki:
https://github.com/avanc/photOS/wiki/Home-Automation

avanc, 2024-10-14:
@tiphie21 Thanks for the good documentation, it was just a cut and paste to add it into the wiki.
@escuco Thanks for the suggestion.

escuco, 2024-11-01:
Hello @tiphie21 

I'm migration my home automation from iobroker to Home Assistant. All my automations are already working except triggering the _photos_ photo frame.

I tried different ways, but I don't know where to manually paste your mentioned action scripts for activating the frame in the new created HA automation.

Would you like to give me a tip?

Thanks
Jan

escuco, 2024-11-01:
@tiphie21 
I found it by myself. I've selected action type "RESTful command" but I had to change your code into:
```
action: rest_command.set_photos_display
metadata: {}
data:
  action: "on"
```
and
```
action: rest_command.set_photos_display
metadata: {}
data:
  action: "off"
```

Now it's working fine.

tiphie21, 2025-01-06:
Hi there,

since I was working on enhance my automation setup, I wanted to also shutdown photOS via HTTP-POST out of my Home Assistant.
So I took my previous code snippets and edited them a little bit:

1. in `configuration.yaml` I added this to my existing `rest_command`:
```
  photos_display_power:
    url: "http://<IP-address>/power"
    method: "POST"
    content_type: "application/x-www-form-urlencoded"
    payload: "action={{ action }}"
```
2. To control it, I used the action data and added this to my automation:
```
action: rest_command.photos_display_power
metadata: {}
data:
  action: shutdown
```

Like @escuco said, one could also just add this into `configuration.yaml`:
```
  photos_display_power:
    url: "http://<IP-address>/power"
    method: "POST"
    content_type: "application/x-www-form-urlencoded"
    payload: "action=shutdown"
```

This could be interesting for someone who want to add this into automations.
Maybe @avanc, you could add this as well to the documentation?