avanc/photOS #128: WiFi wpa_supplicant failed / no link

ueen, 2024-12-25:
After some trouble we got a USB WiFi stick to show as wlan0 and we also get a connection manually setting iwconfig but after a reboot all fail author l without an ethernet connection. 
Please help, wpa_supplicant.conf is no folder or do we need further settings to somehow connect it to wlan0?

avanc, 2025-02-17:
So you got the stick working manually? In that case, the needed kernel modules are available. Might it be that you loaded the drivers manually (e.g. modprobe)? Maybe they have to be defined to be loaded on startup.

avanc, 2025-04-02:
@ueen Are you able to connect to the device via Wifi when you disconnected the LAN? Or is the device just not rebooting?

ueen, 2025-04-02:
Right, with the stick and Lan it works when I remove the Lan (no reboot) whereas a reboot occurs if there's only Lan and I remove it. So my conclusion would be that there is a wlan connection. But it only works when there's a lab connection first.

ueen, 2025-04-02:
Hmm do you have a guide on how to do it? 
Also I noticed the txt work WiFi details or always disappearing from the SD card.