avanc/photOS #132: Random/shuffeling always showing the same?

truidix, 2025-03-19:
Hey,

I really like that project.
Currently we are having around 600 pic and a few videos at the WebDAV share.

Is there an issue that the shuffling is showing very often the same pics?  Because we found that some pics are showed very often and on the other hand some don’t come up.
Does anyone else having that problem? Is there a way to debug that random/shuffling stuff may via activating a specific log? 

Thanks

avanc, 2025-03-20:
The used pseudo random generator provided by Bash might have some limitations, but I really wonder that it is recognizable:
https://ioflood.com/blog/bash-random-number/#Understanding_Randomness_in_Bash

$RANDOM might be replaced by another random source:
https://github.com/avanc/photOS/blob/4c61c70a9dfa5839faaa39bb2ed745ccc12ba19f/package/photoframe/photoframe.sh#L158

to 

    rnd_num=$( python -c "import random; print(random.randint(1,$num_files))" )

See https://www.baeldung.com/linux/bash-draw-random-ints#using-python

It's still a pseudo random generator, but might have better distribution.

@truidix @escuco Can you try to change in /usr/bin/photoframe.sh
I did it on my device, but cannot judge the randomness.

escuco, 2025-03-24:
@avanc @truidix Thanks, I've changed the generator and will see the next days if the images are displayed more randomly.

truidix, 2025-03-24:
Hey @avanc ,

I changed it yesterday. It looks a bit better, as I'm not seeing so mand pics that often.
Even if some videos are still played very often.  

Thanks

Ra5pi, 2025-06-21:
Just like to add that I see the same issue regarding "not so random" shuffling.
Even after changing to the "python" variant I do not see much of an improvement